rfetick/Kalman #3: Use of "inverse" function /// get_sensor_data() missing /// µC blocks

Eheran1, 2021-09-29:
```
Kalman.h:183:16: error: 'class BLA::Matrix<2, 2, BLA::Array<2, 2, float> >' has no member named 'Inverse'
  183 |   K = P*(~H)*S.Inverse(&status);
      |              ~~^~~~~~~
```

The BasicLinearAlgebra Github has [this example](https://github.com/tomstewart89/BasicLinearAlgebra/blob/master/examples/HowToUse/HowToUse.ino) and the syntax is "Invert(Matrix)", so maybe a easy fix? I solved this by changing that section to:

```
  //K = P*(~H)*S.Inverse(&status);
  bool is_nonsingular = Invert(S);
  K = P*(~H)*S;
  if(is_nonsingular){
  //if(!status){
```

Additionally, in the minimal example:
`kalman_minimal:71:9: error: 'get_sensor_data' was not declared in this scope`
Where is this function? Is it like "SIMULATOR_UPDATE()"?

Additionally:
My Feather M0 stops working (needs manual reset into boot loader) when I include this line:
`KALMAN<Nstate,Nobs> K; // your Kalman filter`

FUEL4EP, 2021-10-20:
Thanks a lot, Romain!

Your Kalman library is a great library. I use it successfully on a ATMega1284P MCU for doing online multiple linear regressions. It is astonishing that this is possible with such a small memory footprint on an Arduino.

rfetick, 2022-01-04:
No user reported an error after the patch has been released. I close the issue.

closed 2022-01-04 by rfetick