rfetick/Kalman #8: Examples do not compile

Eheran1, 2023-10-06:
Using "kalman_minimal" example I get tons of error. I tried to hot-fix the issue (array -> matrix etc.) but simply failed.
```
kalman_minimal\kalman_minimal.ino: In function 'void setup()':
kalman_minimal\kalman_minimal.ino:54:22: error: no match for 'operator=' (operand types are 'BLA::Matrix<2, 2, Symmetric<2, float> >' and '<brace-enclosed initializer list>')
            0.0, n2*n2};
                      ^
In file included from BasicLinearAlgebra/BasicLinearAlgebra.h:7:0,
                 from Kalman/Kalman.h:34,
                 from kalman_minimal\kalman_minimal.ino:12:
BasicLinearAlgebra/ElementStorage.h:50:13: note: candidate: BLA::Matrix<Rows, Cols, DType>& BLA::Matrix<Rows, Cols, DType>::operator=(const BLA::Matrix<Rows, Cols, DType>&) [with int Rows = 2; int Cols = 2; DType = Symmetric<2, float>]
     Matrix &operator=(const Matrix &mat)
             ^~~~~~~~
BasicLinearAlgebra/ElementStorage.h:50:13: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const BLA::Matrix<2, 2, Symmetric<2, float> >&'
kalman_minimal\kalman_minimal.ino:57:22: error: no match for 'operator=' (operand types are 'BLA::Matrix<2, 2, Symmetric<2, float> >' and '<brace-enclosed initializer list>')
            0.0, m2*m2};
                      ^
In file included from BasicLinearAlgebra/BasicLinearAlgebra.h:7:0,
                 from Kalman/Kalman.h:34,
                 from kalman_minimal\kalman_minimal.ino:12:
BasicLinearAlgebra/ElementStorage.h:50:13: note: candidate: BLA::Matrix<Rows, Cols, DType>& BLA::Matrix<Rows, Cols, DType>::operator=(const BLA::Matrix<Rows, Cols, DType>&) [with int Rows = 2; int Cols = 2; DType = Symmetric<2, float>]
     Matrix &operator=(const Matrix &mat)
             ^~~~~~~~
BasicLinearAlgebra/ElementStorage.h:50:13: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const BLA::Matrix<2, 2, Symmetric<2, float> >&'
In file included from kalman_minimal\kalman_minimal.ino:12:0:
Kalman/Kalman.h: In instantiation of 'KALMAN<Nstate, Nobs, Ncom, MemF>::KALMAN() [with int Nstate = 2; int Nobs = 2; int Ncom = 0; MemF = BLA::Matrix<2, 2, float>]':
kalman_minimal\kalman_minimal.ino:31:21:   required from here
Kalman/Kalman.h:253:3: error: no matching function for call to 'BLA::Matrix<2, 2, Symmetric<2, float> >::Fill(double)'
   this->P.Fill(0.0);
   ^~~~
In file included from Kalman/Kalman.h:34:0,
                 from kalman_minimal\kalman_minimal.ino:12:
BasicLinearAlgebra/BasicLinearAlgebra.h:78:10: note: candidate: void BLA::MatrixBase<DerivedType, rows, cols, DType>::Fill(const DType&) [with DerivedType = BLA::Matrix<2, 2, Symmetric<2, float> >; int rows = 2; int cols = 2; d_type = Symmetric<2, float>; BLA::MatrixBase<DerivedType, rows, cols, DType>::DType = Symmetric<2, float>]
     void Fill(const DType &val) { *this = val; }
          ^~~~
BasicLinearAlgebra/BasicLinearAlgebra.h:78:10: note:   no known conversion for argument 1 from 'double' to 'const DType& {aka const Symmetric<2, float>&}'
In file included from kalman_minimal\kalman_minimal.ino:12:0:
Kalman/Kalman.h: In instantiation of 'void KALMAN<Nstate, Nobs, Ncom, MemF>::_update(BLA::Matrix<Outputs>, BLA::Matrix<Nstate>) [with int Nstate = 2; int Nobs = 2; int Ncom = 0; MemF = BLA::Matrix<2, 2, float>]':
Kalman/Kalman.h:240:10:   required from 'void KALMAN<Nstate, Nobs, Ncom, MemF>::update(BLA::Matrix<Outputs>) [with int Nstate = 2; int Nobs = 2; int Ncom = 0; MemF = BLA::Matrix<2, 2, float>]'
kalman_minimal\kalman_minimal.ino:75:15:   required from here
[...many more errors...]
```

juanfrapm, 2023-10-25:
Try using other versions of the library BasicLinearAlgebra. I was trying the last one and i got error. Version 3.2 should work!

cosismo, 2023-10-25:
> Try using other versions of the library BasicLinearAlgebra. I was trying the last one and i got error. Version 3.2 should work!

It works for me, changing BasicLinearAlgebra to v 3.2

Thanks!

Riley-Peterson, 2024-08-07:
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: error: no matching function for call to 'BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix()'
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note: candidate: 'template<class ... TAIL> BLA::Matrix<rows, cols, MemT>::Matrix(typename MemT::elem_t, TAIL ...) [with TAIL = {TAIL ...}; int rows = 3; int cols = 3; MemT = Symmetric<3, float>]'
   31 |     Matrix(typename MemT::elem_t head, TAIL... args);
      |     ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note:   template argument deduction/substitution failed:
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: note:   candidate expects at least 1 argument, 0 provided
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix(const BLA::Matrix<3, 3, Symmetric<3, float> >&)'
   12 | class Matrix
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix(BLA::Matrix<3, 3, Symmetric<3, float> >&&)'
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: error: no matching function for call to 'BLA::Matrix<2, 2, Symmetric<2, float> >::Matrix()'
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note: candidate: 'template<class ... TAIL> BLA::Matrix<rows, cols, MemT>::Matrix(typename MemT::elem_t, TAIL ...) [with TAIL = {TAIL ...}; int rows = 2; int cols = 2; MemT = Symmetric<2, float>]'
   31 |     Matrix(typename MemT::elem_t head, TAIL... args);
      |     ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note:   template argument deduction/substitution failed:
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: note:   candidate expects at least 1 argument, 0 provided
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<2, 2, Symmetric<2, float> >::Matrix(const BLA::Matrix<2, 2, Symmetric<2, float> >&)'
   12 | class Matrix
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<2, 2, Symmetric<2, float> >::Matrix(BLA::Matrix<2, 2, Symmetric<2, float> >&&)'
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: error: no matching function for call to 'BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix()'
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note: candidate: 'template<class ... TAIL> BLA::Matrix<rows, cols, MemT>::Matrix(typename MemT::elem_t, TAIL ...) [with TAIL = {TAIL ...}; int rows = 3; int cols = 3; MemT = Symmetric<3, float>]'
   31 |     Matrix(typename MemT::elem_t head, TAIL... args);
      |     ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:31:5: note:   template argument deduction/substitution failed:
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: note:   candidate expects at least 1 argument, 0 provided
  132 | class KALMAN{
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix(const BLA::Matrix<3, 3, Symmetric<3, float> >&)'
   12 | class Matrix
      |       ^~~~~~
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<3, 3, Symmetric<3, float> >::Matrix(BLA::Matrix<3, 3, Symmetric<3, float> >&&)'
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\Kalman/Kalman.h:132:7: error: no matching function for call to 'BLA::Matrix<3, 1, BLA::Array<3, 1, float> >::Matrix()'
  132 | class KALMAN{
      |       ^~~~~~
[... cut ...]
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note: candidate: 'constexpr BLA::Matrix<2>::Matrix(BLA::Matrix<2>&&)'
C:\Users\rpete\OneDrive\Documents\Arduino\libraries\arduino_593447/BasicLinearAlgebra.h:12:7: note:   candidate expects 1 argument, 0 provided

exit status 1

Compilation error: no matching function for call to 'BLA::Matrix<2>::Matrix()'

I am using BasicLinearAlgebra v3.2 and have tried every version and still does not compile the example. Tried in the newer version and older version of arduino ide. There are more errors than just this